ros-visualization/rqt_bag #3: rqt_bag --clock does not work

dirk-thomas, 2017-04-24:
_From @trainman419 on June 6, 2014 1:3_

Playing back a bag file with --clock doesn't work. The clock is only published 2 or 3 times per second, and doesn't appear to increase linearly.

I suspect the issue here is that rospy is using the clock it's publishing and creating some kind of feedback look.

The GUI is also unresponsive while playing back my bag files; this may or may not be related to timer issues caused by using --clock.

_Copied from original issue: ros-visualization/rqt_common_plugins#244_

dirk-thomas, 2017-04-24:
_From @trainman419 on June 6, 2014 17:47_

Did some more testing. rqt_bag publishes and behaves normally when not using `--clock`

dirk-thomas, 2017-04-24:
_From @trainman419 on June 6, 2014 18:7_

I suspect the performance issue here is a result of some very serious lock contention issues, as a result of:
- Each publisher in rqt_bag is handled by its own thread (MessageListenerThread)
- Each of these threads is attempting to publish the clock
- The rospy publisher has an [internal lock](https://github.com/ros/ros_comm/blob/indigo-devel/clients/rospy/src/rospy/topics.py#L817-L825) to make it thread-safe

I'm going to attempt to rewrite the clock publishing so that it listens to the playhead position rather than individual messages, and so that it accepts the `--hz` option that I need for my use case.

dirk-thomas, 2017-04-24:
_From @trainman419 on June 6, 2014 18:43_

More evidence that this is a locking issue: the clock is published monotonically if I use the `--clock` option in conjunction with "Publish All Messages", which forces all messages in the bag file to be published sequentially.

dirk-thomas, 2017-04-24:
_From @bricerebsamen on November 16, 2015 20:22_

beyond the clock problem (that appears to be fixed by #340), rqt_bag fails to publish heavy data (i.e. a bag with data from multiple velodynes) in real time: I get 3Hz per device vs 36Hz with rosbag (it skips a lot of packets)

dirk-thomas, 2017-04-24:
_From @curranw on April 7, 2016 19:55_

I have the same issue, except rqt_bag skips a lot of tf packets, even if the "Play all Messages" option is set. This seems like a serious problem, since it's difficult to visualize a bag if the robot is teleporting around.

If I use rosbag play instead of rqt_bag, everything runs fine.

Is there a current workaround for this?

HWiese1980, 2018-01-25:
Experiencing the same and other issues regarding /clock publishing from rqt_bag. Haven't investigated further yet. But I can confirm that this is a `rqt_bag` issue and that `rosbag` runs just fine.

palikar, 2018-04-24:
I have similar problems. When I start rqt_bag with --clock, open a .bag file and start playing it the /clock is not published at first. When I click in the rqt_bag window, the /clock starts publishing. When the timeline reaches the end and the marker loops back, the /clock stops being published. Then I have to click again in the rqt_bag window in order for the /clock to be published again.