ros-visualization/rqt_bag #7: rqt_bag: plot plugin only lets you plot 1 topic

dirk-thomas, 2017-04-24:
_From @ablasdel on April 21, 2015 3:12_

It would be nice if you could get multiple topics in the same plugin graph for correlation. 

_Copied from original issue: ros-visualization/rqt_common_plugins#312_

dirk-thomas, 2017-04-24:
_From @trainman419 on April 21, 2015 4:48_

The best way to do this would be through the same context menu that allows plotting: instead of just offering to plot a topic, the menu should also allow the user to add the selected topic to an existing plot.

dirk-thomas, 2017-04-24:
_From @ablasdel on April 21, 2015 4:51_

I like that. All we would have to do is give each of the plot windows a unique title and display that as an option in the drop down. This actually doesn't sound super hard at all. If I get some time next week I will see if I can implement this before I release rqt.

dirk-thomas, 2017-04-24:
_From @ablasdel on April 20, 2016 20:1_

Coming up on the self assign anniversary... It is pretty clear other priorities won out. If that looks like it will change I will reassign myself.

VictorLamoine, 2018-04-06:
@dirk-thomas have you worked on this, do you have a prototype code? I'm a python beginner but I need this functionality so I'll try to do this and hopefully make a PR.

dirk-thomas, 2018-04-06:
> have you worked on this, do you have a prototype code?

I haven't. I would have updated the issue if that would be the case.

> I'm a python beginner but I need this functionality so I'll try to fix this issue and hopefully make a PR.

:+1:

lalten, 2018-06-27:
@VictorLamoine Any update on this? :)

VictorLamoine, 2018-06-27:
I did not have time to work on this yet, sorry, feel free to start working on it!