ros-visualization/rqt_bag #9: rqt_bag: Requires core to be open just to examine bag file

dirk-thomas, 2017-04-24:
_From @DLu on August 12, 2015 18:11_

I believe with rxbag you could just open bag files to examine them without a core. However, rqt_bag requires a core, even without trying to publish anything. 

_Copied from original issue: ros-visualization/rqt_common_plugins#332_

dirk-thomas, 2017-04-24:
In rqt the type of a plugin determines if it requires a ROS node or not. The infrastructure currently does not allow to start as non-ROS-node-plugin and then request a ROS node afterwards. Therefore it is currently not possible to implement the requested feature in a single plugin.

I would be possible though to refactor rqt_bag into two plugins: one which does not allow to publish and one with the current functionality. Not a nice approach though.

DLu, 2017-04-24:
I primarily use rqt_bag to see the timeline of a bagfile and inspect the contents. I rarely use it for playing back the messages. Other people's mileage may vary. 

I think the ideal solution would be a command line flag that specifies whether you will want to publish or not. I'm unclear if that would still require two plugins.

sloretz, 2025-06-02:
ROS 1 is end-of-life (EOL) as May 31st 2025, so I'm closing this.

[If you still rely on ROS 1, read this page to learn about your options.](https://bit.ly/NoeticEOL)

closed 2025-06-02 by sloretz