ros-visualization/rqt_bag #12: Desktop launcher

VictorLamoine, 2017-05-30:
I think it would be nice to have a Ubuntu desktop launcher so that an end user can easily launch `rqt_bag`, open his bag file and play with the data.

Currently I'm using a script + desktop launcher:

`cat /usr/local/bin/rqt_bag_desktop` (make sure it has execute rights)
```bash

#!/bin/bash
source /opt/ros/kinetic/setup.bash

RESULT=$(pgrep roscore && echo Running)

if [ "${RESULT:-null}" = null ]; then
  echo "roscore not running! Running it as background process"
  roscore &
  sleep 0.5
fi

rqt_bag
```

`cat /usr/local/share/applications/rqt_bag.desktop`
```bash
[Desktop Entry]
Name=ROS Qt bag
Comment=Record and play bag files
Exec=/usr/local/bin/rqt_bag_desktop
Icon=/usr/share/icons/Adwaita/scalable/apps/utilities-system-monitor-symbolic.svg
Type=Application
Categories=ROS;
StartupNotify=true
```

Is there a way to include such a thing in this package?

- We need to make sure the chosen icon exists on the end user system
- Should we close `roscore` when the application exit?

dirk-thomas, 2017-07-07:
One problem with this goal is that the ROS Debian packages are installed into `/opt/ros/...` rather than `/usr`. So any boundled `.desktop` file wouldn't be usable out-of-the-box. And the package can't install those files into `/usr` since there might be multiple `rqt_bag` packages from different ROS distributions. Therefore I don't think this can be achieve.

However a `.desktop` could be installed to `/opt/ros/...` in order to serve as a template users can copy to the right location if they want to.

Regarding the icon it would probably be best to embed an icon file into the package to ensure it is there.

Regarding the start / stop of `roscore` I am not sure that it is a good idea. It is pretty opaque to the user and either way will be "wrong" for some use cases. Letting the user start the core explicitly is the way to go with the least surprise imo.

dirk-thomas, 2018-03-21:
I will close the ticket for now since there was no further feedback. Please feel free to comment on the closed ticket and if necessary it can be reopened.

closed 2018-03-21 by dirk-thomas

VictorLamoine, 2018-03-22:
At least this issue may help other users achieve the same functionality.

I understood what you said and I agree, it's not straightforward to do this because it will only work in specific cases.