ros-visualization/rqt_bag #15: NameError: global name 'qInfo' is not defined

mikepurvis, 2017-10-24:
Issue appears in Lunar on OS X, see https://github.com/mikepurvis/ros-install-osx/issues/89.

Two instances, both in `bag_widget.py`: https://github.com/ros-visualization/rqt_bag/search?q=qInfo

closed 2017-10-24 by dirk-thomas

yosoufe, 2017-11-01:
Hi,
When I am trying to open the `rqt_bag` it gives me this error?

mikepurvis, 2017-11-01:
Per [this page](http://doc.qt.io/qt-5/qtglobal.html#qInfo), `qInfo` was introduced in QT 5.5, and Xenial has 5.5.1:

https://packages.ubuntu.com/source/xenial/qtbase-opensource-src

I wonder if this is possibly a gap in Xenial's version of pyqt5? Noting that QtCore comes in via that:

https://github.com/ros-visualization/python_qt_binding/search?q=QtCore

dirk-thomas, 2017-11-01:
> Could either switch these back to qWarns, or put a polyfill in our python_qt_binding package to supply the missing function as required.

`qWarning` implies printing to `stderr` which doesn't seem appropriate. I would propose to switch the calls to `qDebug` instead. See #17.

I don't see what `python_qt_binding` should map the call to?

mikepurvis, 2017-11-01:
Yeah, that's not really an answer. Good call to just switch it to debug.

dirk-thomas, 2017-11-01:
@yosoufe `rqt_bag` version 0.4.11 will fix the problem. I have just released that version so it might be a bit until it is available from Debian packages. Sorry for the inconvenience.

yosoufe, 2017-11-21:
Thanks @dirk-thomas How can I check my rqt_bag version? Because I still have the problem. I also checked with my colleagues and they have the same problem. Do you know how long it may take until the debian repo get updated?

dirk-thomas, 2017-12-03:
> How can I check my rqt_bag version?

`dpkg -l | grep rqt-bag`

> Do you know how long it may take until the debian repo get updated?

The "release manager" for each ROS distro makes the decision when to sync the latest packages from the testing repo to the stable repo. You can follow the conversation on [discourse](https://discourse.ros.org/t/preparing-for-kinetic-sync-2017-11-29/3275) for updates on the progress.

PR #16 (2017-10-24): Import qInfo function
Regression introduced in #14, closes #15 and closes https://github.com/mikepurvis/ros-install-osx/issues/89.