ros-visualization/rqt_bag #22: Segmentation fault ROS Kinetic

jokla, 2017-11-19:
Hi!

I have the version 0.4.11-0xenial-20171101-091232-0800 of rqt_bag and I get a segmentation fault if i try to run it. 

`user@IDG2033:~$ rqt_bag
Segmentation fault (core dumped)
`

The problem seems related to PyQt4.QtGui 

Any tips?

Thanks!

dirk-thomas, 2017-12-03:
Please provide more information which allows to reproduce the problem. I will close the ticket for now but once you have commented with the necessary information it can be reopened.

closed 2017-12-03 by dirk-thomas

iliis, 2017-12-06:
I had the same issue. I'm not sure what exactly fixed it but it seems to be some mismatch between [py]qt versions. Try to uninstall [py]qt4 and install [py]qt5 (for python2!).

VictorLamoine, 2018-05-07:
@dirk-thomas please re-open

Here is a trace:
```gdb
$ gdb -ex r --args python ./install/bin/rqt_bag
GNU gdb (Ubuntu 7.11.1-0ubuntu1~16.5) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from python...Reading symbols from /usr/lib/debug/.build-id/66/b65551cc62504bf758e2f57514d33cf3017b8f.debug...done.
done.
Starting program: /usr/bin/python ./install/bin/rqt_bag
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffe06df700 (LWP 24864)]
[New Thread 0x7fffd6f80700 (LWP 24865)]
[New Thread 0x7fffd677f700 (LWP 24866)]
[New Thread 0x7fffd5f7e700 (LWP 24867)]
[New Thread 0x7fffc57b9700 (LWP 24869)]
[New Thread 0x7fffc4fb8700 (LWP 24870)]
[New Thread 0x7fffb75a7700 (LWP 24872)]
[New Thread 0x7fffb6da6700 (LWP 24873)]
[New Thread 0x7fffb65a5700 (LWP 24876)]
[New Thread 0x7fffb5da4700 (LWP 24879)]
[Thread 0x7fffb5da4700 (LWP 24879) exited]
[Thread 0x7fffc57b9700 (LWP 24869) exited]
[New Thread 0x7fffc57b9700 (LWP 24880)]
[New Thread 0x7fffb5da4700 (LWP 24881)]
[New Thread 0x7fffb49b9700 (LWP 24882)]
[New Thread 0x7fff9d40c700 (LWP 24883)]
[New Thread 0x7fff9cc0b700 (LWP 24884)]
[New Thread 0x7fff9a40a700 (LWP 24885)]
[New Thread 0x7fff95c09700 (LWP 24886)]
[New Thread 0x7fff93408700 (LWP 24887)]
[New Thread 0x7fff90c07700 (LWP 24888)]
[New Thread 0x7fff8e406700 (LWP 24889)]

Thread 1 "python" received signal SIGSEGV, Segmentation fault.
0x00007fff88c1e5a4 in ?? () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
(gdb) bt
#0  0x00007fff88c1e5a4 in ?? () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#1  0x00007ffff7de76ba in call_init (l=<optimized out>, argc=argc@entry=2, argv=argv@entry=0x7fffffffd928, env=env@entry=0x2496920) at dl-init.c:72
#2  0x00007ffff7de77cb in call_init (env=0x2496920, argv=0x7fffffffd928, argc=2, l=<optimized out>) at dl-init.c:30
#3  _dl_init (main_map=main_map@entry=0x1120d50, argc=2, argv=0x7fffffffd928, env=0x2496920) at dl-init.c:120
#4  0x00007ffff7dec8e2 in dl_open_worker (a=a@entry=0x7fffffff85b0) at dl-open.c:575
#5  0x00007ffff7de7564 in _dl_catch_error (objname=objname@entry=0x7fffffff85a0, errstring=errstring@entry=0x7fffffff85a8, mallocedp=mallocedp@entry=0x7fffffff859f, 
    operate=operate@entry=0x7ffff7dec4d0 <dl_open_worker>, args=args@entry=0x7fffffff85b0) at dl-error.c:187
#6  0x00007ffff7debda9 in _dl_open (file=0x115d6e0 "/usr/lib/python2.7/dist-packages/PyQt4/QtGui.so", mode=-2147483646, caller_dlopen=0x51ad19 <_PyImport_GetDynLoadFunc+233>, nsid=-2, 
    argc=<optimized out>, argv=<optimized out>, env=0x2496920) at dl-open.c:660
#7  0x00007ffff75ecf09 in dlopen_doit (a=a@entry=0x7fffffff87e0) at dlopen.c:66
#8  0x00007ffff7de7564 in _dl_catch_error (objname=0xa53570, errstring=0xa53578, mallocedp=0xa53568, operate=0x7ffff75eceb0 <dlopen_doit>, args=0x7fffffff87e0) at dl-error.c:187
#9  0x00007ffff75ed571 in _dlerror_run (operate=operate@entry=0x7ffff75eceb0 <dlopen_doit>, args=args@entry=0x7fffffff87e0) at dlerror.c:163
#10 0x00007ffff75ecfa1 in __dlopen (file=<optimized out>, mode=<optimized out>) at dlopen.c:87
#11 0x000000000051ad19 in _PyImport_GetDynLoadFunc () at ../Python/dynload_shlib.c:140
#12 0x000000000051a8e4 in _PyImport_LoadDynamicModule () at ../Python/importdl.c:42
#13 0x00000000004a4ae1 in import_submodule.lto_priv () at ../Python/import.c:2722
#14 0x00000000004a7fd5 in ensure_fromlist.lto_priv () at ../Python/import.c:2628
#15 0x00000000004a3f38 in import_module_level.isra.3 (level=0, fromlist=('QtGui', 'QtCore', 'uic'), globals=<optimized out>, name=<optimized out>) at ../Python/import.c:2291
#16 PyImport_ImportModuleLevel () at ../Python/import.c:2310
#17 0x00000000004a59e4 in builtin___import__ () at ../Python/bltinmodule.c:49
[... 185 lines cut ...]
#180 0x00000000004b9ab6 in PyEval_EvalCodeEx () at ../Python/ceval.c:3582
#181 0x00000000004c16e7 in fast_function (nk=<optimized out>, na=<optimized out>, n=<optimized out>, pp_stack=0x7fffffffd2e0, func=<function at remote 0x7fffe46b6938>) at ../Python/ceval.c:4445
#182 call_function (oparg=<optimized out>, pp_stack=0x7fffffffd2e0) at ../Python/ceval.c:4370
#183 PyEval_EvalFrameEx () at ../Python/ceval.c:2987
#184 0x00000000004b9ab6 in PyEval_EvalCodeEx () at ../Python/ceval.c:3582
#185 0x00000000004c16e7 in fast_function (nk=<optimized out>, na=<optimized out>, n=<optimized out>, pp_stack=0x7fffffffd4e0, func=<function at remote 0x7fffe46b6a28>) at ../Python/ceval.c:4445
#186 call_function (oparg=<optimized out>, pp_stack=0x7fffffffd4e0) at ../Python/ceval.c:4370
#187 PyEval_EvalFrameEx () at ../Python/ceval.c:2987
#188 0x00000000004b9ab6 in PyEval_EvalCodeEx () at ../Python/ceval.c:3582

#189 0x00000000004eb30f in PyEval_EvalCode (
[... cut ...]
#192 0x00000000004e3cd6 in PyRun_SimpleFileExFlags () at ../Python/pythonrun.c:948
#193 0x0000000000493ae2 in Py_Main () at ../Modules/main.c:640
#194 0x00007ffff7810830 in __libc_start_main (main=0x4934c0 <main>, argc=2, argv=0x7fffffffd928, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffd918)
    at ../csu/libc-start.c:291
#195 0x00000000004933e9 in _start ()
```

- Ubuntu 16.04, ROS Kinetic
- `rqt_bag` from latest git.
- Crashes when `python-pyqtgraph` is installed, works fine when it is removed.

dirk-thomas, 2018-05-07:
@VictorLamoine The stacktrace shows that the `rqt_bag` executable is using Qt 4 (`libQtGui.so.4`). Kinetic only support Qt 5. That is why it fails when it dynamically loads the Qt 4 libraries. `python-pyqtgraph` is using Qt 4 and seems to make the Qt 4 libraries available *before* the Qt 5 libraries. I don't know what `rqt` could do about this.