ros-visualization/rqt_bag #23: ImportError: cannot import name qInfo

AbnerCSZ, 2017-12-20:
```
abner@abner-XPS-8700:~/bagfiles$ rqt_bag nsh_indoor_outdoor.bag 
Traceback (most recent call last):
  File "/opt/ros/kinetic/bin/rqt_bag", line 5, in <module>
    from rqt_bag.bag import Bag
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_bag/bag.py", line 39, in <module>
    from .bag_widget import BagWidget
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_bag/bag_widget.py", line 40, in <module>
    from python_qt_binding.QtCore import Qt, qInfo, qWarning, Signal

```

My ros is kinetic and system is Ubuntu 16.04.
I remember that I had encountered before, but also solved. I modified a few words of a file about QT to fix the error. But now I can not find the URL for that solution, and I forget how to fix it.

dirk-thomas, 2017-12-20:
Duplicate of #16.

closed 2017-12-20 by dirk-thomas

AbnerCSZ, 2017-12-20:
How can I update version?
```
abner@abner-XPS-8700:~$ sudo apt-get upgrade ros-kinetic-rqt-bag
Reading package lists... Done
Building dependency tree       
Reading state information... Done
ros-kinetic-rqt-bag is already the newest version (0.4.10-0xenial-20171025-113423-0800).
Calculating upgrade... Done
```

dirk-thomas, 2017-12-20:
The new version 0.4.11 should already be available for Ubuntu Xenial (see http://repositories.ros.org/status_page/ros_kinetic_default.html?q=rqt_bag). So the standard `sudo apt update && sudo apt dist-upgrade` should achieve that.

AbnerCSZ, 2017-12-20:
I had sudo apt update && sudo apt dist-upgrade.
But system says that my rqt_bag is the newest. 
```
abner@abner-XPS-8700:~$ sudo apt-get upgrade ros-kinetic-rqt-bag
Reading package lists... Done
Building dependency tree       
Reading state information... Done
**ros-kinetic-rqt-bag is already the newest version (0.4.10-0xenial-20171025-113423-0800).**
Calculating upgrade... Done
```

dirk-thomas, 2017-12-20:
You might want to check your apt configuration since the official apt repo does contain the newer version as shown by the status page as well as can be seen http://packages.ros.org/ros/ubuntu/dists/xenial/main/binary-amd64/Packages and http://packages.ros.org/ros/ubuntu/pool/main/r/ros-kinetic-rqt-bag/

AbnerCSZ, 2017-12-20:
Thank you very much.