ros-visualization/rqt_bag #27: Does not work on Melodic

VictorLamoine, 2018-06-05:
- Kubuntu 18.04
- ROS Melodic

```bash
$ dpkg -l | grep rqt_bag
ii  ros-melodic-rqt-bag                             0.4.12-0bionic.20180522.045045              amd64        rqt_bag provides a GUI plugin for displaying and replaying ROS bag files.
ii  ros-melodic-rqt-bag-plugins                     0.4.12-0bionic.20180522.162523              amd64        rqt_bag provides a GUI plugin for displaying and replaying ROS bag files.
```

I can open a bag file but I cannot see any contents inside:

```python
Exception in thread Thread-7:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rqt_bag/timeline_cache.py", line 74, in run
    msg_stamp, item = self.loader(topic, stamp, item_details)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rqt_bag_plugins/image_timeline_renderer.py", line 160, in _load_thumbnail
    msg_topic, msg, msg_stamp = bag._read_message(pos)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rosbag/bag.py", line 1299, in _read_message
    return self._reader.seek_and_read_message_data_record(position, raw)
TypeError: seek_and_read_message_data_record() takes exactly 4 arguments (3 given)

Exception in thread Thread-15:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rqt_bag/message_loader_thread.py", line 83, in run
    msg_data = self._get_message(bag, playhead_position)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rqt_bag/message_loader_thread.py", line 96, in _get_message
    msg_data = self.timeline.read_message(bag, position)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rqt_bag/bag_timeline.py", line 481, in read_message
    return bag._read_message(position)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rosbag/bag.py", line 1299, in _read_message
    return self._reader.seek_and_read_message_data_record(position, raw)
TypeError: seek_and_read_message_data_record() takes exactly 4 arguments (3 given)

```

In the GUI:
- The thumbnails do not work
- The topics cannot be displayed as raw (the created panel stays empty)
- The topics cannot be displayed as images (the created panel stays empty)
- Plotting the data crashes `rqt_bag`

VictorLamoine, 2018-06-05:
I don't understand this error as the definition only seems to take 3 arguments:
https://github.com/ros/ros_comm/blob/melodic-devel/tools/rosbag/src/rosbag/bag.py#L2358

Can someone confirm this bug?

lalten, 2018-06-08:
I have the exact same problem, with both the `0.4.12-0bionic.20180522.045045` debian package and the `0.4.12` git master. Did you find a solution?

lalten, 2018-06-08:
Apparently ros_comm [PR #1372](https://github.com/ros/ros_comm/pull/1372) broke this. If you revert commit [`752cf1f8bf3a17560f5d496609fadab2ebffd121`](https://github.com/ros/ros_comm/commit/752cf1f8bf3a17560f5d496609fadab2ebffd121) in ros_comm, rqt_bag works as expected again.

dirk-thomas, 2018-08-03:
Closing as duplicate of ros/ros_comm#1432.

closed 2018-08-03 by dirk-thomas