ros-visualization/rqt_bag #29: read bag bug?

zezhou, 2018-08-07:
rqt_bag occurs a error when I open some bags, but it can work well in other bags.
here is the error info:

Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_bag/bag_timeline.py", line 491, in on_mouse_down
    TimelinePopupMenu(self, event, topic)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_bag/timeline_menu.py", line 213, in __init__
    self.process(action)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_bag/timeline_menu.py", line 262, in process
    frame.show(self.timeline.get_context())
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_bag/timeline_menu.py", line 85, in show
    self._viewer = self._viewer_type(self._timeline, self, self._topic)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_bag_plugins/plot_view.py", line 91, in __init__
    self.plot_widget = PlotWidget(timeline, parent, topic)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_bag_plugins/plot_view.py", line 176, in __init__
    msg = bag._read_message(entry.position)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosbag/bag.py", line 986, in _read_message
    return self._reader.seek_and_read_message_data_record(position, raw)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosbag/bag.py", line 2475, in seek_and_read_message_data_record
    msg_type = _get_message_type(connection_info)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosbag/bag.py", line 1571, in _get_message_type
    message_type = genpy.dynamic.generate_dynamic(info.datatype, info.msg_def)[info.datatype]
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genpy/dynamic.py", line 168, in generate_dynamic
    tmp_file.file.write(full_text.encode())
UnicodeDecodeError: 'ascii' codec can't decode byte 0xe8 in position 574: ordinal not in range(128)

dirk-thomas, 2018-08-07:
This looks like a duplicate of ros/genpy#94.

zezhou, 2018-08-09:
yes, it looks like the same problem. Close this thread.

closed 2018-08-09 by zezhou