ros-visualization/rqt_bag #31: rqt_bag has no vertical scroll bar until window is resized

jaredkhan, 2018-11-13:
I am on Ubuntu 16.04 and running rqt_bag with a large bag. The list of topics is taller than the window so a vertical scroll bar is very helpful. This scrollbar only shows up if I resize the window after loading.

[image]

kmhallen, 2018-12-19:
Confirmed in Ubuntu 18.04 ROS Melodic too

ElonKou, 2019-03-28:
I have the same questioon.
Have you solved it yet?

erichahn, 2020-05-26:
Bump. Still a present issue.

closed 2020-10-20 by mjeronimo

marko-curlin, 2022-04-02:
I have this issue on Ubuntu 20.04 ROS Noetic. I open a bag with a big number of topics, and only after resizing or moving the window does the scroll bar appear in its place.

PR #63 (2020-10-20): Fix issue: no vertical scroll bar until window is resized
Previously, when loading a bag, if the new timeline height was larger
than the bag widget window height, the scroll bar would not show up until
the window was resized by the user. This fix forces a resize upon
opening a bag so that the new timeline height is taken into account
and the scrollbar appears if the timeline height is now greater than the
bag widget height, as expected. 

Fixes #31