ros-visualization/rqt_bag #35: Python3 Compatibility Variable Type long

tommy3001, 2019-01-06:
There is an issue in 
rqt_bag/plugins/raw_view.py
and
rqt_bag_plugins/plot_view.py
regarding PEP 0237.
Python 3 handles _long_ data types as an _int_,  _long_ is therefore not available:

`
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python3.7/site-packages/rqt_bag/plugins/message_view.py", line 92, in event
    self.message_viewed(bag, msg_data)
  File "/opt/ros/melodic/lib/python3.7/site-packages/rqt_bag/plugins/raw_view.py", line 65, in message_viewed
    self.message_tree.set_message(msg)
  File "/opt/ros/melodic/lib/python3.7/site-packages/rqt_bag/plugins/raw_view.py", line 103, in set_message
    self._add_msg_object(None, '', '', msg, msg._type)
  File "/opt/ros/melodic/lib/python3.7/site-packages/rqt_bag/plugins/raw_view.py", line 186, in _add_msg_object
    if type(obj) in [int, long, float]:
NameError: name 'long' is not defined
`

tommy3001, 2019-01-06:
I deleted long from the list in the regarding files. For me it is working now....

dirk-thomas, 2019-01-07:
Please consider to provide a pull request for this. The same issue has been address in other packages, e.g.: https://github.com/ros/ros_comm/blob/fe9479cdbf0be0caa542c74c7c1fb8229ea8164d/tools/rostopic/src/rostopic/__init__.py#L67-L70

closed 2020-08-07 by dirk-thomas

eric1221bday, 2020-08-19:
@dirk-thomas #51 only fixes one file with this issue I believe, this issue still persists in `raw_view.py`. I'll see if I can get another PR to fix that, but in the meantime we might want to keep this open.

PR #51 (2020-08-07): py3 fix: long = int
fixes #35