ros-visualization/rqt_bag #37: Topic parameter 'data_class' is not initialized

ElonKou, 2019-03-28:
Hi,
When I use rqt_bag to record a topic, some of the topics cannot be recorded and a warning is displayed, but when I use the command line tool rosbag, these topics can be recorded.
I am working under ubuntu16.04

> [WARN] [1553769837.934367]: Error subscribing to /huace/nmea_sentence (ignoring): topic parameter 'data_class' is not initialized
[WARN] [1553769838.267172]: Error subscribing to /lslidar_sweep (ignoring): topic parameter 'data_class' is not initialized
[WARN] [1553769838.271660]: Error subscribing to /lslidar_packet0 (ignoring): topic parameter 'data_class' is not initialized
[WARN] [1553769838.277671]: Error subscribing to /tong/nmea_sentence (ignoring): topic parameter 'data_class' is not initialized
[WARN] [1553769838.285201]: Error subscribing to /lslidar_packet1 (ignoring): topic parameter 'data_class' is not initialized

I checked the corresponding code and found that the function could not get their correct type for the topic of the previous problem.
[pytype = roslib.message.get_message_class(datatype)](https://github.com/ros-visualization/rqt_bag/blob/master/rqt_bag/src/rqt_bag/recorder.py#L162)

Can you help me determine what is going on in the end, how to solve it, thank you very much.

Guozhongyuan, 2023-10-19:
Because rospy cannot find the msg description.

You can copy msg file and import it,
```
import os
import sys
sys.path.append('./devel/lib/python3/dist-packages')
from your_msgs.msg import *
```

Or another method is to `source devel/setup.bash` before run python file.

sloretz, 2025-06-02:
ROS 1 is end-of-life (EOL) as May 31st 2025, so I'm closing this.

[If you still rely on ROS 1, read this page to learn about your options.](https://bit.ly/NoeticEOL)