ros-visualization/rqt_bag #38: rqt_bag doesn't publish all messages in the bag file

meghakadole, 2019-04-01:
I have to replay a bag file, which contains 15000 messages over a duration of 66seconds. I wrote a subscriber node to this topic. On the subscriber side, I count the number of messages received. I notice that not all messages are received. And the number of messages received is not fixed every time. I need to replay this bag file in real-time and without data loss. How can I achieve this?

I tried replaying the same bag with rosbag play (command line) the performance of command line is much better when compared to rqt_bag, but there is still some data loss. How can I ensure that all messages are received on subscriber end? 
 Your help is much appreciated. Thanks in advance.

ScottYang1982, 2020-09-13:
I've same issue , anyone can give some hints, thanks~

kekeliu-whu, 2021-10-30:
I have similar problem: when opening a bag file([link](https://drive.google.com/file/d/1bwXvlEX0RYkixR3LDHte-U_KpMMIRG8x/view?usp=sharing)) with 500hz imu data, only 100hz is shown in rqt_bag

kekeliu-whu, 2021-10-30:
You can try to increase your subscriber queue size.

kekeliu-whu, 2021-10-30:
I have 2 similar problems when opening a bag file ([link](https://drive.google.com/file/d/1bwXvlEX0RYkixR3LDHte-U_KpMMIRG8x/view)) with 500hz imu data:
1. Only 100hz is shown in rqt_bag;
2. When i play topic /imu_correct in the rqt_bag, `rosbag hz /imu_correct` show that the frequency is only 25hz.

nguyenlam185, 2022-02-10:
I am facing the same issue here.
I have a bag file which contains over 80,000 messages, duration is about 40s. The `rqt_bag` can only publish the topics at ~50Hz whereas the `rosbag play` command can do that at ~2000Hz.
Are there any parameters that need to be set before starting the `rqt_bag` in order to speed it up?