ros-visualization/rqt_bag #74: CompressedImage thumbnails not displayed in Noetic

mabelzhang, 2020-10-27:
The error is printed from a call to `rqt_bag_plugins.image_helper.imgmsg_to_pil`, but I don't know if the cause is in this package or somewhere deeper in the actual image conversion, because I see different behavior in Melodic and Noetic while compiling from the same commit hash.

Version: commit hash 3beb083ac2988246f5c7ff579dc4bbacff8d339c
Platforms: Melodic vs. Noetic

To reproduce:
In `rqt_bag`, load a rosbag containing message type `sensor_msgs/CompressedImage`. Toggle thumbnail on.

In Melodic, image thumbnails are displayed. No messages are printed in the terminal.

In Noetic, no image thumbnails are displayed. These messages are printed in the terminal
```
Can't convert image: initial_value must be str or None, not bytes
Disabling renderer on /camera/color/image_raw/compressed
```

mjeronimo, 2020-11-11:
I've tried the provided bag file with Kinetic, Melodic, and Noetic and the image_helper patch works for all of them. The thumbnail shows correctly as well as the images when using the Image View plugin. 

Without this PR, rqt_bag fails on all three distributions with the following messages:

* **Kinetic** - fails with "Can't convert image: local variable 'pil_mode' referenced before assignment
* **Melodic** - fails with "Can't convert image: local variable 'pil_mode' referenced before assignment
* **Noetic** - fails with "Can't convert image: initial_value must be str or None, not bytes 

To test, I did the following:

_(All console windows source /opt/ros/\<distribution\>/setup.bash in the .bashrc)_
1. In one console window, start roscore

// Try the distribution's rqt_bag
2. Open a console window
3. Run rqt_bag
4. Open the images_compressed.bag file
5. Select Thubnail. No thumbnails are displayed and there is an error message in the console window
6. Select View|Image and select the Next button on the image plugin. No images are displayed and an error message is displayed in the console window

// Try the latest rqt_bag
7. Open another console window
8. Build the 'master' branch of latest rqt_bag source code (I used colcon)
9. Source rqt_bag/install/local_setup.bash
10. Run rqt_bag (the newly built one which will be available after sourcing local_setup.bash. You can confirm with "which rqt_bag")
11. Repeat steps 3-6 above. This time thumbnails and images are displayed correctly and there are no error messages

The above is true for all of Kinetic, Melodic, and Noetic. Please let me know if you get a different result following these steps.

sloretz, 2025-06-02:
ROS 1 is end-of-life (EOL) as May 31st 2025, so I'm closing this.

[If you still rely on ROS 1, read this page to learn about your options.](https://bit.ly/NoeticEOL)

closed 2025-06-02 by sloretz