ros-visualization/rqt_bag #90: Timeline not displayed due to no '_mergesort' function

ktkt2020, 2020-12-08:
It seems the `_mergesort` function in the `rosbag.bag` was replaced by `heapq.merge` (https://github.com/ros/ros_comm/pull/2017).
But in the current version https://github.com/ros-visualization/rqt_bag/blob/b36db71b0af957853a113b9336f5929a5c4d7fc3/rqt_bag/src/rqt_bag/bag_timeline.py#L273
and
https://github.com/ros-visualization/rqt_bag/blob/b36db71b0af957853a113b9336f5929a5c4d7fc3/rqt_bag/src/rqt_bag/bag_timeline.py#L303
the old function was still called.
This might cause the timeline to show incorrectly.

zgxsin, 2020-12-18:
I am facing the same issue here.

BurryChen, 2020-12-24:
I have the same issue.

ktkt2020, 2020-12-28:
Adding the original `_mergesort` function back to `rosbag/bag.py` could temporarily help to show the Timeline (root permission required for the apt-installed ROS). Perhaps an adaptation of `heapq.merge` might be required for ROS1's `rqt_bag`.

https://github.com/ros/ros_comm/blob/47ad91df5cf0a60f29140be21e78b1587d0adf98/tools/rosbag/src/rosbag/bag.py#L2906:L2968

MichaelGrupp, 2021-02-02:
I created a pull request with a fix: #93

commit b0e0b5e2 (2021-02-11): Fix crash during merge sort of bag entries (fixes #90) (#93)

As noted in #90, the previously used function was removed in ros_comm.

This makes rqt_bag usable again on ROS Noetic and is compatible with
Python 2 distros.
closed 2021-02-11 by mjeronimo