ros-visualization/rqt_bag #102: ROS 2 version of rqt_bag has missing icons

clalancette, 2021-04-28:
Prior to https://github.com/ros-visualization/qt_gui_core/pull/250 being merged, icons didn't work at all on Linux.  Once that PR is merged, icons work for the most part, but there are a few missing in rqt_bag (see the attached file for a screenshot).

[image]

From the tooltips, these missing icons look like "Zoom In", "Zoom Out", "Zoom Home", and "Toggle Thumbnails"

emersonknapp, 2022-09-01:
This is related to https://github.com/ros-visualization/rqt/issues/249 - though from what I can tell, the Tango theme doesn't provide the icons that are missing:

* `zoom-in`
* `zoom-out`
* `zoom-original`
* `insert-image`

I don't know the historical context - maybe original rqt apps were depending on Adwaita theme instead? This seems to have those icons under those exact names, at least in its "legacy" listings.

emersonknapp, 2022-09-02:
Oh - this issue is true on Noetic, too. I think it's an Ubuntu Focal+ problem, not a ROS2 problem.

ahcorde, 2025-11-12:
[image]

I can see the icons on `rolling`. Closing this one