ros-visualization/rqt_bag #103: ROS2 version of rqt_bag will not publish any data

mschuckmann, 2021-09-24:
Using ROS2 Galactic I can not get the rqt_bag player to publish topics from a recorded bag file. 
I've right clicked on on each topic in the player timeline and checked "Publish" but nothing works. 
I've been testing with the attached turtlesim bag file. 

[rosbag2_2021_09_24-11_25_26.zip](https://github.com/ros-visualization/rqt_bag/files/7227815/rosbag2_2021_09_24-11_25_26.zip)

mjeronimo, 2021-09-25:
I just tried the bag file on Galactic using rqt_bag and playback worked fine. Silly question, but did you press Play after checking Publish?

mschuckmann, 2021-09-25:
Hi mjeronimo, thanks for the reply, I certainly hope the problem is something silly that I am doing but I don’t think so as both I and another co-worker are having the same problem. 
I did click play I also tried stepping through the file and seeking to multiple points. Note that the issue is with Publishing data as in ros_bag sending messages to topics, how did you go about checking for published data? , I’m using `ros2 topic echo …` and I’m not seeing anything.

mjeronimo, 2021-09-27:
@mschuckmann  I used ros2 topic echo as well for the two topics. I ran this using the latest Galactic binaries: https://github.com/ros2/ros2/releases/download/release-galactic-20210716/ros2-galactic-20210616-linux-focal-amd64.tar.bz2.