ros-visualization/rqt_bag #108: Running on ROS2 Rolling Jammy

ijnek, 2022-03-31:
The following error occurs when running

* ``ros2 run rqt_bag rqt_bag`` or
* ``rqt`` and adding the rqt bag widget manually

```
PluginManager._load_plugin() could not load plugin "rqt_bag/Bag":
Traceback (most recent call last):
  File "/opt/ros/rolling/lib/python3.10/site-packages/qt_gui/plugin_handler.py", line 102, in load
    self._load()
  File "/opt/ros/rolling/lib/python3.10/site-packages/qt_gui/plugin_handler_direct.py", line 55, in _load
    self._plugin = self._plugin_provider.load(self._instance_id.plugin_id, self._context)
  File "/opt/ros/rolling/lib/python3.10/site-packages/qt_gui/composite_plugin_provider.py", line 72, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/rolling/lib/python3.10/site-packages/qt_gui/composite_plugin_provider.py", line 72, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/rolling/lib/python3.10/site-packages/rqt_gui_py/ros_py_plugin_provider.py", line 69, in load
    return super(RosPyPluginProvider, self).load(plugin_id, ros_plugin_context)
  File "/opt/ros/rolling/lib/python3.10/site-packages/qt_gui/composite_plugin_provider.py", line 72, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/rolling/lib/python3.10/site-packages/rqt_gui/ros_plugin_provider.py", line 107, in load
    return class_ref(plugin_context)
  File "/home/ijnek/Downloads/tmp/soccer_vision_msgs_layers_ws/build/rqt_bag/src/rqt_bag/bag.py", line 58, in __init__
    self._widget = BagWidget(context, args.clock)
  File "/home/ijnek/Downloads/tmp/soccer_vision_msgs_layers_ws/build/rqt_bag/src/rqt_bag/bag_widget.py", line 84, in __init__
    self._timeline = BagTimeline(context, publish_clock)
  File "/home/ijnek/Downloads/tmp/soccer_vision_msgs_layers_ws/build/rqt_bag/src/rqt_bag/bag_timeline.py", line 111, in __init__
    self._timeline_frame = TimelineFrame(self)
  File "/home/ijnek/Downloads/tmp/soccer_vision_msgs_layers_ws/build/rqt_bag/src/rqt_bag/timeline_frame.py", line 131, in __init__
    self._topic_font.setPointSize(self._topic_font_size)
TypeError: setPointSize(self, int): argument 1 has unexpected type 'float'
```

Environment:
```
OS: Ubuntu22.04 Jammy
Distro: Rolling
ROS Installation Method: Binary
```

This is a detailed example for https://github.com/ros-visualization/rqt/issues/266. Although this can be resolved in rqt_bag by ensuring the argument types are correct, the issue may arise across any library that depends on ``python_qt_binding``.

ivanpauno, 2022-04-19:
https://github.com/ros-visualization/rqt_bag/pull/109 should've fixed the issue.
A new release is needed to get the issue fixed when installing from debians.

closed 2022-04-19 by ivanpauno

ijnek, 2022-04-20:
I think #109 has enough changes to get rqt_bag up and running, but there are more changes required to prevent runtime errors. Some of those are covered in https://github.com/ijnek/rqt_bag/commit/a0286d91f9e4618c0eeb0f03299625b7134766a9 and involve data types being casted to the expected type before being passed to methods.
The reason I haven't raised a PR is that I haven't tried going through rqt_bag to identify all problems. I can't guarantee that there are loads of other issues.

Could we either re-open or open another issue to track these?

geoeo, 2022-05-13:
I installed ros2 rolling via the apt (which comes with rqt_bag) and its definitely not fixed.

```PluginManager._load_plugin() could not load plugin "rqt_bag/Bag":
Traceback (most recent call last):
  File "/opt/ros/rolling/local/lib/python3.10/dist-packages/qt_gui/plugin_handler.py", line 102, in load
    self._load()
  File "/opt/ros/rolling/local/lib/python3.10/dist-packages/qt_gui/plugin_handler_direct.py", line 55, in _load
    self._plugin = self._plugin_provider.load(self._instance_id.plugin_id, self._context)
  File "/opt/ros/rolling/local/lib/python3.10/dist-packages/qt_gui/composite_plugin_provider.py", line 72, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/rolling/local/lib/python3.10/dist-packages/qt_gui/composite_plugin_provider.py", line 72, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/rolling/lib/python3.10/site-packages/rqt_gui_py/ros_py_plugin_provider.py", line 69, in load
    return super(RosPyPluginProvider, self).load(plugin_id, ros_plugin_context)
  File "/opt/ros/rolling/local/lib/python3.10/dist-packages/qt_gui/composite_plugin_provider.py", line 72, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/rolling/lib/python3.10/site-packages/rqt_gui/ros_plugin_provider.py", line 107, in load
    return class_ref(plugin_context)
  File "/opt/ros/rolling/lib/python3.10/site-packages/rqt_bag/bag.py", line 58, in __init__
    self._widget = BagWidget(context, args.clock)
  File "/opt/ros/rolling/lib/python3.10/site-packages/rqt_bag/bag_widget.py", line 84, in __init__
    self._timeline = BagTimeline(context, publish_clock)
  File "/opt/ros/rolling/lib/python3.10/site-packages/rqt_bag/bag_timeline.py", line 111, in __init__
    self._timeline_frame = TimelineFrame(self)
  File "/opt/ros/rolling/lib/python3.10/site-packages/rqt_bag/timeline_frame.py", line 131, in __init__
    self._topic_font.setPointSize(self._topic_font_size)
TypeError: setPointSize(self, int): argument 1 has unexpected type 'float'

```

clalancette, 2022-05-27:
I've opened #111 to track the ongoing work.

PR #109 (2022-04-19): Fix the types being passed into QFont and QColor.
PyQt 5.15 in Ubuntu Jammy is a lot more picky about the types
it accepts for QFont.setFontSize() and the QColor() constructor.
Use the types it expects here, which makes this at least launch
under Ubuntu Jammy.

This fixes #108