ros-visualization/rqt_bag #111: Running rqt_bag against PyQt 5.15.6 can still crash

clalancette, 2022-05-27:
The version of PyQt on Ubuntu 22.04 is 5.15.6, and it is a lot more picky about the types being passed into API calls.  In particular, trying to pass in float values to API calls like `setPointSize(self, int)` result in an exception now.  We've fixed some of these, but more need to be fixed.  See #108 for more information.

geoeo, 2022-07-02:
Is there an ETA on this? This is a big problem for image based workflows

ijnek, 2022-07-13:
@geoeo

The runtime errors are hard to find (and time consuming), I believe the maintainers (or myself) aren't going to get around to this soon.

I think the quickest way to solve this would be to squash one error at a time as they arise. Would you be able to post the python stack traces in this thread?

ijnek, 2022-07-15:
@geoeo That exact error you posted earlier should have been resolved in [this rolling sync](https://discourse.ros.org/t/new-packages-for-ros-2-rolling-ridley-2022-05-26/25786/1) a while back. Do you have some other errors that you could post here?

geoeo, 2022-07-16:
@ijnek Is this part of humble as well or is it planned to be?

geoeo, 2022-07-16:
I just updated my rolling and humble via ubuntu software updater. 
And now I get a different error

```
  File "/opt/ros/humble/lib/python3.10/site-packages/rqt_bag/timeline_frame.py", line 297, in paint
    self._draw_time_divisions(painter)
  File "/opt/ros/humble/lib/python3.10/site-packages/rqt_bag/timeline_frame.py", line 651, in _draw_time_divisions
    self._draw_major_divisions(painter, major_stamps, start_stamp, major_division)
  File "/opt/ros/humble/lib/python3.10/site-packages/rqt_bag/timeline_frame.py", line 673, in _draw_major_divisions
    painter.drawText(label_x, label_y, label)
TypeError: arguments did not match any overloaded call:
  drawText(self, Union[QPointF, QPoint], str): argument 1 has unexpected type 'float'
  drawText(self, QRectF, int, str): argument 1 has unexpected type 'float'
  drawText(self, QRect, int, str): argument 1 has unexpected type 'float'
  drawText(self, QRectF, str, option: QTextOption = QTextOption()): argument 1 has unexpected type 'float'
  drawText(self, QPoint, str): argument 1 has unexpected type 'float'
  drawText(self, int, int, int, int, int, str): argument 1 has unexpected type 'float'
  drawText(self, int, int, str): argument 1 has unexpected type 'float'
Traceback (most recent call last):
  File "/opt/ros/humble/lib/python3.10/site-packages/rqt_bag/timeline_frame.py", line 297, in paint
    self._draw_time_divisions(painter)
  File "/opt/ros/humble/lib/python3.10/site-packages/rqt_bag/timeline_frame.py", line 651, in _draw_time_divisions
    self._draw_major_divisions(painter, major_stamps, start_stamp, major_division)
  File "/opt/ros/humble/lib/python3.10/site-packages/rqt_bag/timeline_frame.py", line 673, in _draw_major_divisions
    painter.drawText(label_x, label_y, label)
TypeError: arguments did not match any overloaded call:
  drawText(self, Union[QPointF, QPoint], str): argument 1 has unexpected type 'float'
  drawText(self, QRectF, int, str): argument 1 has unexpected type 'float'
  drawText(self, QRect, int, str): argument 1 has unexpected type 'float'
  drawText(self, QRectF, str, option: QTextOption = QTextOption()): argument 1 has unexpected type 'float'
  drawText(self, QPoint, str): argument 1 has unexpected type 'float'
  drawText(self, int, int, int, int, int, str): argument 1 has unexpected type 'float'
  drawText(self, int, int, str): argument 1 has unexpected type 'float'

```

The error is the same in both humble and rolling

ijnek, 2022-07-17:
> @ijnek Is this part of humble as well or is it planned to be?

Yes, such bug ports are/will be backported to humble.
In regards to your last error, i have addressed the issues in #118 and raised a PR.

geoeo, 2022-07-17:
Do you know when that pull request will be released on humble? Are they monthly/quarterly releases?

ijnek, 2022-07-17:
Once this PR is merged, the package maintainer would have to backport the changes to the humble branch, make a new release, and then that will get synced to the distros with no specific schedule, but they happen usually once every 1-2 months.

If you want to get notified about the releases, you could enable notifications for the [release tag on ROS 
Discourse](https://discourse.ros.org/c/release/16).

In the meantime, there's nothing stopping you from building this repo from source (once the PR is merged and backported to Humble) and using it in your projects!