ros-visualization/rqt_bag #119: Fix ros2 and rolling branch not in sync

ivanpauno, 2022-07-21:
https://github.com/ros-visualization/rqt_bag/tree/ros2 and https://github.com/ros-visualization/rqt_bag/tree/rolling should be synced.

ivanpauno, 2022-07-21:
FYI @audrow.
Is it fine if I force push `ros2` into `rolling` and the cherry-pick https://github.com/ros-visualization/rqt_bag/commit/5048c25ffc0baeb5658b6986298ab4c7ecb4808a on top again?

audrow, 2022-07-22:
I believe that the SHA1 needs to be the same for both branches. I think the best thing would be to force push ros2 onto rolling.

One thing to note is that both branches need the same SHA1, so I don't think cherry-picking is the way to go.

ivanpauno, 2022-07-22:
Fixed.

I thought https://github.com/ros-visualization/rqt_bag/commit/5048c25ffc0baeb5658b6986298ab4c7ecb4808a was missing in the `ros2` branch, but it was already there.

closed 2022-07-22 by ivanpauno