ros-visualization/rqt_bag #127: ros2: recording doesn't create metadata.yaml

emersonknapp, 2022-10-07:
`rqt_bag` recordings leave the bag without `metadata.yaml` file, which makes it unusable by some tooling and requires a `ros2 bag reindex`

Andre-ex9, 2023-05-02:
I'm facing the same issue. Is it a bug or is this the way rqt_bag is intended to work ?

emersonknapp, 2023-05-03:
This is a bug - I think that the recorder/writer isn't being deconstructed properly so it doesn't create the file.

Yadunund, 2023-05-18:
Managed to save the metadata.yaml file when closing the plugin with this PR https://github.com/ros-visualization/rqt_bag/pull/141
After closing rqt_bag, I was able to `ros2 bag play BAG` and echo the topics in another terminal.

The "Save bag" button is still disabled. Looking into that next.

Yadunund, 2023-05-19:
Will keep this open until https://github.com/ros-visualization/rqt_bag/pull/141 is merged

clalancette, 2024-05-09:
We merged in #141 ages ago, so closing this out.

closed 2024-05-09 by clalancette