ros-visualization/rqt_bag #136: ros2: rqt_bag shows only one data of multiple topics

nonanonno, 2022-12-04:
I have a rosbag file which has two topics: `/chatter` (`std_msgs/msg/String`) and `/number` (`std_msgs/msg/Int32`). But when I show it in rqt_bag, there is a case that displays only `/chatter` data for both `/chatter` and `/number` like the following figure.

[image]

I think this happens when the timestamp of two topics are exactly same. In this case, `/chatter` is 1FPS and `/number` is 2FPS. So if I seek timeline to time only `/number` is publishing, I can see both `/chatter` and `/number` correctly.

## Enviornment

- Ubuntu 22.04
- ROS2 Humble
- rqt_bag version: 1.1.4

nonanonno, 2022-12-04:
>  when the timestamp of two topics are exactly same

This is because I have used python script which uses rosbag2_py.SequentialWriter to create rosbag file.

guihomework, 2023-04-21:
I confirm this bug, but I recorded using `ros2 bag record /imu /clock`  on Ubuntu 22.04 ros rolling (19 April download) using rqt_bag-1.3.1/rqt_bag_plugins-1.3.1

and `rqt_bag -v rosbag2_2023_04_21-09_39_18` to visualize
both raw show imu data and not the clock although I requested a display of clock first (was already showing imu data)

[image]

verbose output
```
PluginManager._discover() using cached plugin discovery information
QtBindingHelper using pyqt
PerspectiveManager.switch_perspective() switching to perspective "@rqt_bag__Bag"
MainWindow.restore_settings()
PluginManager.restore_settings_without_plugins()
MainWindow.restore_state()
PluginManager.load_plugin(rqt_bag/Bag, 0)
MainWindow.save_setup()
RosPyPluginProvider._init_node() initialize ROS node "rqt_gui_py_node_1911"
Start called on RclpySpinner, spinning ros2 node
RospkgPluginProvider._find_plugins() crawling for plugins of type 'rqt_bag'
rqt_bag.TimelineFrame.load_plugins() loaded plugin "rqt_bag_plugins/BagImagePlugin"
rqt_bag.TimelineFrame.load_plugins() loaded plugin "rqt_bag_plugins/BagPlotPlugin"
PluginManager._load_plugin_restore()
PluginManager._load_plugin(rqt_bag/Bag#0) successful
Loading 'b'rosbag2_2023_04_21-09_39_18'' ...
Loading bag from metadata file "rosbag2_2023_04_21-09_39_18" Succeeded
Topics from this bag: ['/clock', '/imu']
PluginHandler.restore_settings()
PluginHandler.emit_restore_settings_completed()
PluginManager._emit_load_plugin_completed()
Setting timeline left: 1682062761.1051958 right: 1682062767.0358496
Setting playhead Time(nanoseconds=1682062761105195776, clock_type=SYSTEM_TIME)
MainWindow.restore_state()
Done loading 'b'rosbag2_2023_04_21-09_39_18''
Setting timeline left: 1682062761.1051958 right: 1682062767.0358496
press, rel=PyQt5.QtCore.QPoint(268, 87), global=PyQt5.QtCore.QPoint(535, 1360), diff=PyQt5.QtCore.QPoint(535, 1360)
press, rel=PyQt5.QtCore.QPoint(265, 88), global=PyQt5.QtCore.QPoint(532, 1361), diff=PyQt5.QtCore.QPoint(532, 1361)
TimelinePopupMenu process action <PyQt5.QtWidgets.QAction object at 0x7fd55c6ead40>
Adding listener to topic /clock of type <class 'rqt_bag.plugins.raw_view.RawView'>
TimelinePopupMenu process action <PyQt5.QtWidgets.QAction object at 0x7fd596864670>
Adding listener to topic /imu of type <class 'rqt_bag.plugins.raw_view.RawView'>
```

berndpfrommer, 2023-12-19:
I'm seeing the same bug with two image topics that have identical time stamps. It's very confusing.

Yadunund, 2024-02-14:
Also happening on Iron as reported https://github.com/ros-visualization/rqt_bag/issues/147

arneboe, 2024-02-14:
This happens to me for all messages, not just with the same timestamp.
I attached a debugger an can see that the ListenerEvent is posted correctly in message_listener_thread.py:82.
But the event is received by all views. I.e. a view for topic "a" will also receive the messages for topic "b".
I have no idea why that happens

berndpfrommer, 2024-12-16:
Hit this bug again today. It makes rqt_bag useless when dealing with synchronized data. Could somebody please fix this?

PhilippSchmaelzle, 2026-10-01:
Hello, I would like to add a new version of this bug. At least it looks similar.
Just if someone sees similar behavior.

**Setup:**
I record a `ìmage_raw` and the corresponding `camera_info` live in a vehicle.
So I have `image_raw + camera_info` pairs with exact same `header.stamp`.
Afterwards I wanted to *reorder* the bag so the bag log time is rewritten to the messages *header.stamp*.

**Problem:**
When I looked at the result I saw duplicate `camera_info` messages. So the exact same message was twice in the bag - at least it looked like it since I used *rqt_bag*.
But a simple *ros2 bag play + ros2 topic echo* showed no duplicates.

**Investigation:**
I found this issue and tried the following:
- misuse the `camera_info.height` field as a sequence number
- add 1ms offset to the new *log time*. So `log_time = header.stamp + 1ms`

And voila, no more duplicates show up in *rqt_bag*.

The attached images show this behavior. 
Each image shows two *rqt_bag* GUI where the upper one is the corrupt example and the lower one shows the same stamp/message but with the manipulated *log time*.

You need to focus on the `header.stamp` and the `log time` in the bottom row of the *rqt_bag* GUI and the fake *sequence number* in the `height` field helps as well to show, that in the corrupt case twice the same message is shown.

---
### t0
[image]

---
### t0+1
[image]