ros-visualization/rqt_bag #146: Can't play bags with `v9` metadata

Fixit-Davide, 2023-12-21:
Since https://github.com/ros2/rosbag2/pull/1476 there may be an incompatibility with the current capabilities of this RQT plugin.

I get this error when trying to load a bag to play:
```
Loading `bag/file/path` failed due to: Exception on parsing info file: yaml-cpp: error at line 11, column 11: bad conversion
```

CC @roncapat

roncapat, 2023-12-22:
@MichaelOrlov I tested again and it is something happening in:
```
            self.reader = rosbag2_py.SequentialReader()
            self.reader.open(
                rosbag2_py.StorageOptions(uri=bag_path), rosbag2_py.ConverterOptions())
```
so related to rosbag2. Any clue?

Line 11 column 11 is:
```yaml
rosbag2_bagfile_information:
  version: 9
  storage_identifier: sqlite3
  duration:
    nanoseconds: 352757433264
  starting_time:
    nanoseconds_since_epoch: 1703179683403182547
  message_count: 161592
  topics_with_message_count:
    - topic_metadata:
        name: /zed_camera/odom
        type: nav_msgs/msg/Odometry
        serialization_format: cdr
        offered_qos_profiles:
          - history: unknown    # THIS ONE
            depth: 0
            reliability: reliable
            durability: volatile
            deadline:
              sec: 9223372036
              nsec: 854775807
            lifespan:
              sec: 9223372036
              nsec: 854775807
            liveliness: automatic
            liveliness_lease_duration:
              sec: 9223372036
              nsec: 854775807
            avoid_ros_namespace_conventions: false
        type_description_hash: RIHS01_3cc97dc7fb7502f8714462c526d369e35b603cfc34d946e3f2eda2766dfec6e0
      message_count: 5939
```

I think there is an issue in YAML compatibility of the `metadata.yaml`... why that line starts as a list item?

roncapat, 2023-12-22:
I validated the aboe YAML via: https://onlineyamltools.com/validate-yaml
and converted to JSON with: https://onlineyamltools.com/convert-yaml-to-json
because I had a doubt that `offered_qos_profiles` had a syntax issue, but I got what it seems to me a reasonable confirmation of YAML getting parsed correctly (I added some comments to get them RED when rendered by github):
```json
{
  "rosbag2_bagfile_information": {
    "version": 9,
    "storage_identifier": "sqlite3",
    "duration": {
      "nanoseconds": 352757433264
    },
    "starting_time": {
      "nanoseconds_since_epoch": 1703179683403182600
    },
    "message_count": 161592,
    "topics_with_message_count": [
      {
        "topic_metadata": {
          "name": "/zed_camera/odom",
          "type": "nav_msgs/msg/Odometry",
          "serialization_format": "cdr",
          "offered_qos_profiles": [   ### BEGIN LIST
            {                         ### BEGIN QOS #1 ITEM
              "history": "unknown",
              "depth": 0,
              "reliability": "reliable",
              "durability": "volatile",
              "deadline": {
                "sec": 9223372036,
                "nsec": 854775807
              },
              "lifespan": {
                "sec": 9223372036,
                "nsec": 854775807
              },
              "liveliness": "automatic",
              "liveliness_lease_duration": {
                "sec": 9223372036,
                "nsec": 854775807
              },
              "avoid_ros_namespace_conventions": false
            }                         ### END QOS #1 ITEM
          ],                          ### END LIST
          "type_description_hash": "RIHS01_3cc97dc7fb7502f8714462c526d369e35b603cfc34d946e3f2eda2766dfec6e0"
        },
        "message_count": 5939
      }
    ]
  }
}
```

roncapat, 2023-12-22:
After a depper inspection, this was a local issue on one of my machines. Closing.

MichaelOrlov, 2023-12-22:
@roncapat Thanks for investigating this issue.
For future reference. The `"offered_qos_profiles"` is a list because it could be multiple publishers with different QoS profiles on the same topic and we store all of them.

roncapat, 2023-12-22:
Thanks! I had more a doubt with the appearance of the list itself / was not confident whether that was valid YAML syntax. Anyway, better safe than sorry :) I'm starting to use it on real hardware on real projects, so will do some useful testing before Jazzy will be branched out !

BTW, happy New Year :)

closed 2023-12-22 by Fixit-Davide