ros-visualization/rqt_bag #147: ros2 Iron: rqt_bag displays wrong type in raw view

tcm0116, 2024-01-19:
When displaying the raw view for multiple topics in a bag file, rqt_bag always uses the same message type for all raw views:

[image]

This can be recreated by:

Terminal 1:
1. `source /opt/ros/iron/setup.bash`
2. `ros2 run turtlesim turtlesim_node`

Terminal 2:
1. `source /opt/ros/iron/setup.bash`
2. `ros2 bag record -a -o rosbag`
3. After several seconds, press `Ctrl+C` to stop the recording
4. `rqt_bag rosbag`
5. Right-click on `/turtle1/color_sensor` and select View and then select Raw
6. Right-click on `/turtle1/pose` and select View and then select Raw
7. Select the Next frame button in the main toolbar until data is displayed in the raw views
8. The message type and data layout in both raw views will be `turtlesim/msg/Pose`
9. Select the Next frame button in the `/turtle1/color_sensor__Raw` view
10. The message type and data layout in both raw views will be `turtlesim/msg/Color`
11. Select the Next frame button in the `/turtle1/pose__Raw` view
12. The message type and data layout in both raw views will be `turtlesim/msg/Pose`

arneboe, 2024-02-14:
I am having the same issue in ros2 iron.
Any updates on this?

Yadunund, 2024-02-14:
I'lll close this ticket since it's a duplicate of https://github.com/ros-visualization/rqt_bag/issues/136

closed 2024-02-14 by Yadunund