ros-visualization/rqt_bag #155: Im trying to play bag a bag in mcap format and rqt crashes.

r4hul77, 2024-04-12:
I'm trying to play bag a ros2 bag which is in mcap format it breaks with the following. Looks like rqt bag is thinking this is in sql3 format. Is there anyway to fix it ?
```
Traceback (most recent call last):
  File "/opt/ros/humble/lib/python3.10/site-packages/rqt_bag/bag_widget.py", line 287, in _handle_load_clicked
    self.load_bag(filename)
  File "/opt/ros/humble/lib/python3.10/site-packages/rqt_bag/bag_widget.py", line 324, in load_bag
    self._timeline.add_bag(bag)
  File "/opt/ros/humble/lib/python3.10/site-packages/rqt_bag/bag_timeline.py", line 176, in add_bag
    self._timeline_frame.reset_timeline()
  File "/opt/ros/humble/lib/python3.10/site-packages/rqt_bag/timeline_frame.py", line 974, in reset_timeline
    self.playhead = Time(seconds=self._stamp_left)
  File "/opt/ros/humble/lib/python3.10/site-packages/rqt_bag/timeline_frame.py", line 236, in _set_playhead
    bag, entry = self.scene().get_entry(self._playhead, topic)
  File "/opt/ros/humble/lib/python3.10/site-packages/rqt_bag/bag_timeline.py", line 320, in get_entry
    bag_entry = bag.get_entry(t, topic)
  File "/opt/ros/humble/lib/python3.10/site-packages/rqt_bag/rosbag2.py", line 134, in get_entry
    result = self._execute_sql_query(sql_query, topic)
  File "/opt/ros/humble/lib/python3.10/site-packages/rqt_bag/rosbag2.py", line 170, in _execute_sql_query
    entries = cursor.execute(base_query + sql_query).fetchall()
sqlite3.DatabaseError: file is not a database
```

Yadunund, 2024-04-13:
Looks like you've sourced a ROS 2 Humble workspace. `mcap` is the default bag format from ROS 2 Iron so cli tools like `ros2 bag` will assume the bag format is `sqlite3` which is the default in Humble. Either source Iron, or you could try pass a storage config file as cli arg. For more information see `ros2 bag -h` 
```
  --storage-config-file STORAGE_CONFIG_FILE
                        Path to a yaml file defining storage specific configurations. See storage plugin documentation for
                        the format of this file.
 ```
 
 I'll close this ticket but feel free to reopen if you find an issue with Iron or the storage config file.

closed 2024-04-13 by Yadunund

TannerGilbert, 2025-06-09:
I'm trying to use MCAP with ROS 2 Humble and running into the same issue with `rqt_bag`. `ros2 bag` detects `mcap` as expected but `rqt_bag` doesn't and non of the command line arguments allow for changing the bag format as far as I can see.