ros-visualization/rqt_bag #166: rqt_bag slow/unusable on iron/jazzy

befinitiv, 2024-06-06:
Hi

Coming from melodic, I always enjoyed analyzing bag data using rqt_bag tool. Recently, I started a new project using ROS2 iron and also jazzy and I noticed that the rqt_bag tool is almost unusable for me, quite in contrast to the melodic version.

Situation:
* A bag in either sqlite or MCAP format
* Contents:
```Files:             rosbag2_2024_06_05-13_57_47_0.mcap
Bag size:          2.4 GiB
Storage id:        mcap
Duration:          59.399s
Start:             Jun  5 2024 13:57:47.539 (1717588667.539)
End:               Jun  5 2024 13:58:46.939 (1717588726.939)
Messages:          15982
Topic information: Topic: /diagnostics | Type: diagnostic_msgs/msg/DiagnosticArray | Count: 59 | Serialization Format: cdr
                   Topic: /rosout | Type: rcl_interfaces/msg/Log | Count: 45 | Serialization Format: cdr
                   Topic: /sensors/camera0/camera_info | Type: sensor_msgs/msg/CameraInfo | Count: 2254 | Serialization Format: cdr
                   Topic: /sensors/camera0/image_raw/ffmpeg | Type: ffmpeg_image_transport_msgs/msg/FFMPEGPacket | Count: 2250 | Serialization Format: cdr
                   Topic: /sensors/gnss0/fix | Type: sensor_msgs/msg/NavSatFix | Count: 514 | Serialization Format: cdr
                   Topic: /sensors/gnss0/ntrip_client/rtcm | Type: rtcm_msgs/msg/Message | Count: 16 | Serialization Format: cdr
                   Topic: /sensors/gnss0/ubx_nav_cov | Type: ublox_ubx_msgs/msg/UBXNavCov | Count: 512 | Serialization Format: cdr
                   Topic: /sensors/gnss0/ubx_nav_hp_pos_llh | Type: ublox_ubx_msgs/msg/UBXNavHPPosLLH | Count: 512 | Serialization Format: cdr
                   Topic: /sensors/gnss0/ubx_nav_pvt | Type: ublox_ubx_msgs/msg/UBXNavPVT | Count: 512 | Serialization Format: cdr
                   Topic: /sensors/gnss0/ubx_nav_status | Type: ublox_ubx_msgs/msg/UBXNavStatus | Count: 102 | Serialization Format: cdr
                   Topic: /sensors/gnss0/ubx_rxm_rawx | Type: ublox_ubx_msgs/msg/UBXRxmRawx | Count: 1177 | Serialization Format: cdr
                   Topic: /sensors/gnss0/ubx_rxm_rtcm | Type: ublox_ubx_msgs/msg/UBXRxmRTCM | Count: 60 | Serialization Format: cdr
                   Topic: /sensors/gnss0/ubx_rxm_sfrbx | Type: ublox_ubx_msgs/msg/UBXRxmSfrbx | Count: 975 | Serialization Format: cdr
                   Topic: /sensors/ouster0/imu | Type: sensor_msgs/msg/Imu | Count: 5849 | Serialization Format: cdr
                   Topic: /sensors/ouster0/metadata | Type: std_msgs/msg/String | Count: 1 | Serialization Format: cdr
                   Topic: /sensors/ouster0/nearir_image | Type: sensor_msgs/msg/Image | Count: 585 | Serialization Format: cdr
                   Topic: /sensors/ouster0/points | Type: sensor_msgs/msg/PointCloud2 | Count: 553 | Serialization Format: cdr
                   Topic: /tf_static | Type: tf2_msgs/msg/TFMessage | Count: 6 | Serialization Format: cdr
```
So it is neither small nor big. Using melodic rqt_bag I worked in the past on bags much much larger without any issues.

* System: Intel i7-5600U 12GB RAM, Ubuntu 22.04

Symptoms are:
* Even after the bag has loaded completely (indicated by the progress bar at the bottom), rqt_bag occupies the CPU 100% for a couple of minutes, then idles (only true for iron)
* The red marker cannot be repositioned, it stays at the beginning of the bag
* Sometimes (no obvious explanation why) the marker can be moved, but only moves after a delay of several seconds

Obviously, with these issues the tool is not usable.

Since I haven't found other people having similar issues, I am wondering whether there is something I am doing wrong here. Any advice would be highly appreciated.

FedeAi, 2024-07-03:
Same for me on Ros Iron

soncininicolas, 2024-12-26:
Having a similar issue for ROS Humble. We used it all the time in ROS1 and now with the same robot and even smaller bag sizes we get huge wait times, impossibility to operate with the QT window, or application crashes.
It has made it impossible to explore our data with it and we'll have to look for alternatives :'(

peci1, 2025-05-07:
>   * The red marker cannot be repositioned, it stays at the beginning of the bag
>   * Sometimes (no obvious explanation why) the marker can be moved, but only moves after a delay of several seconds

These are fixed by #177.

peci1, 2025-05-08:
Also, #178 greatly improves load times (but doesn't fix the slowness after the bag is loaded).